logisim-evolution/logisim-evolution #2209: Custom appearance does not always get saved

slyofzero, 2025-04-20:
I am using Logoism Evolution version 3.9 and again facing the exact same issues as in https://github.com/logisim-evolution/logisim-evolution/issues/1379.

Step 1 - Create a new subcircuit
Step 2 - Edit the subcircuit appearance for a custom one
Step 3 - Go back to the main circuit and drag and drop the created subcircuit
Step 4 - The custom appearance doesn't show instead the same default one does

[image]

[image]

I am new to logism evolution so I might be missing something, but I tried creating a custom subcircuit apperance in just logism and didn't face this issue. I read something about setting appearance to custom, but couldn't find the dropdown menu option for that.

davidhutchens, 2025-04-20:
The Appearance attribute is in the Properties panel on the left. In your last picture it is at the bottom of the list. But you are looking at the instance of the halfAdder in the main circuit. That isn't where you should fix it. To fix all of the instances of the circuit, open the subcircuit (your halfAdder), make sure nothing is selected. Do not view the custom circuit. The panel should have the title `Circuit: halfAdder`. Change the Appearance to Custom there and it should apply to all instances of halfAdder.

maehne, 2025-04-20:
I am closing this as a duplicate of issue #1780. Let's continue the discussion there.

closed 2025-04-20 by maehne