logisim-evolution/logisim-evolution #2212: win 10 - Error writing to file: ucrtbase.dll

wanghuijing, 2025-04-25:
Subject: Installation Error on Windows 11 (Logisim Evolution 3.9.0)

Dear Logisim Support Team,

I’m encountering an installation issue with Logisim-evolution-3.9.0-x86_64.msi on my system:

OS: Windows 11 (x64)

Java Version: 21 (installed)

Previous Version: Logisim 2.7.1 (used before)

Error Message:

Error writing to file:  
D:\software\Logisim\3.9-release\runtime\bin\ucrtbase.dll  
Verify that you have access to that directory.  
Troubleshooting Steps Taken:

Granted full permissions (including "Modify/Write" access) to the target folder.

Ran the installer as Administrator (same error persists).

Question:
Could this be related to Java 21 compatibility, residual files from v2.7.1, or a known issue with the MSI installer? Any guidance would be appreciated.

Thank you!

Mudassir10X, 2025-04-26:
This issue is most probably caused due to permission setting and not the java or msi installer as I have used it on many systems with the same JAVA and windows. Please ensure that the permissions you give to the target folder are not only for your user but to system and administrator. The Windows file permission system is very hard to deal with. if you still cannot get you head around using the GUI, using MS-DOS cmd or powershell and checking permissions using the "dir" command might help you to understand what user does not have the permission.

maehne, 2025-04-26:
If you use [the native installer for your OS](https://github.com/logisim-evolution/logisim-evolution#download), you don't need to install a separate JDK. The path `D:\software\Logisim\3.9-release\`, to which you try to install Logisim-evolution, is not the standard path for software installation on Windows. Therefore, it is up to you to ensure that the installing user has the proper access rights to the installation destination. If the issue persists, I recommend to the latest [nightly build](https://github.com/logisim-evolution/logisim-evolution#nightly-builds).

closed 2025-05-27 by maehne