logisim-evolution/logisim-evolution #2224: Gradle task createMsi fails with Eclipse Temurin JDK 24 on Windows

maehne, 2025-05-02:
Trying to build an MSI package using Eclipse Temurin JDK 24 on Windows 11 Pro 24H2 (x86_64) yields:

```
./gradlew createMsi
Picked up JAVA_TOOL_OPTIONS: -Duser.language=en
Starting a Gradle Daemon, 1 incompatible and 4 stopped Daemons could not be reused, use --status for details

> Task :genVhdlSyntax
Picked up JAVA_TOOL_OPTIONS: -Duser.language=en

> Task :createMsi FAILED
Picked up JAVA_TOOL_OPTIONS: -Duser.language=en
jlink failed with: Error: This JDK does not contain packaged modules and cannot be used to create another image with the jdk.jlink module
Command "C:\Program Files\Eclipse Adoptium\jdk-24.0.0.36-hotspot/bin/jpackage" failed with RC 1.

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':createMsi'.
> Error while creating the MSI package.

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.

BUILD FAILED in 34s
7 actionable tasks: 7 executed
```

Creating an MSI package works using Eclipse Temurin JDK 23 and Azul Zulu JDK 24.

This issue seems to be related to https://github.com/adoptium/adoptium-support/issues/1267, [JDK-8349648](https://bugs.openjdk.org/browse/JDK-8349648), and [this Eclipse Temurin blog post on "Eclipse Temurin JDK 24.0.0+36 enables JEP 493"](https://adoptium.net/zh-CN/blog/2025/03/eclipse-temurin-jdk24-JEP493-enabled/). This [discussion on the impact of JEP 493](https://www.ej-technologies.com/blog/2025/04/cross-platform-jre-bundle-creation-under-threat-from-jep-493/) confirms that Eclipse Temurin JDK 24 fully adopts JEP 493, whereas Azul Zulu and other JDK distributions continue to provide the `.jmod` files.

Even though, I have so far only observed this issue on Windows, I fear that also other tasks relying on `jlink` and `jpackage` to create distribution packages might be affected by JEP 493 and thus break once it gets fully adopted by other JDK distributions on Linux and macOS.

VSENTHAMIZHAN, 2025-05-02:
@maehne Hi maintainers,

I would like to work on resolving Issue #2224 related to incorrect or misaligned label placement in specific circuit layouts within Logisim Evolution.

maehne, 2025-05-02:
> [@maehne](https://github.com/maehne) Hi maintainers,
> 
> I would like to work on resolving Issue [#2224](https://github.com/logisim-evolution/logisim-evolution/issues/2224) related to incorrect or misaligned label placement in specific circuit layouts within Logisim Evolution.

@VSENTHAMIZHAN: Thanks for volunteering! However, do you actually mean this issue about the failing `createMsi` task or another issue, which you describe with "related to incorrect or misaligned label placement in specific circuit layouts within Logisim Evolution"? Please, always post your comments to the respective issue to keep discussion focused!

maehne, 2025-05-02:
So, I added `-add modules java.base,java.compiler,java.desktop,java.naming,java.prefs,java.scripting,java.sql` to the `params` list in [line 157](https://github.com/logisim-evolution/logisim-evolution/blob/aeae0825b02177c1eb5162a2f1f7c7e591792b43/build.gradle.kts#L157) of `build.gradle.kts`, which gets passed to `jpackage`. With this, `createMsi` task works again, and the installed size of Logisim-evolution decreases considerably from 184 MB to 106 MB! A quick test of the installed application suggests that it is still working, including the help viewer.

maehne, 2025-05-05:
Closing issue as resolved by PR #2225. Follow-up issue is #2227 for further improving the Gradle build scripts.

davidhutchens, 2025-05-06:
> By the way, does anyone have an idea, why we have two very similarly named JARs in the build/libs folder?

@maehne: You asked about the small .jar file in build/libs. That file seems to be built by shadow. Even if I make a regular jar with `./gradlew jar`, doing `./gradlew shadowJar` will replace the regular jar with the small one while creating the shadowJar. I don't know how to have it not do that.

maehne, 2025-05-07:
Thanks for your feedback to my questions regarding the two created JARs in `build/libs` I suggest that we track this problem in a new issue. The small JAR is empty except for a nearly empty MANIFEST so that I think it can't be crucial to the plugin's correct operation. I suspect that it has more something to do with us customizing the name of the output JAR.