logisim-evolution/logisim-evolution #2239: A memory component's output should not be initialized as "U" (undefined)

prlaba, 2025-05-21:
The output of a memory component should not be initialized as "U" (undefined) or "E" (error).  It should be random, either 0 or 1.

For example, the D flip-flop component is initialized with its Q output "U" and its /Q output "E".  The Q output should be initialized randomly to 0 or 1 and its /Q output initialized to the inverse of Q. This matches the behavior of a real D flip-flop.

davidhutchens, 2025-05-21:
I can reproduce this with the D flip-flop in both 3.9.0 and head of main. The issue does imply the selection of the random start state option for memory. Otherwise, it is initialized as expected to 0.

prlaba, 2025-05-21:
The behavior I described is with the project option _Memory elements start in unknown/random state (rather than 0)_ checked.

I'm running L-E v3.9.0.

BFH-ktt1, 2025-05-27:
The choice for "U"/"E" for the flipflops was selected due to the behavior observed in HDL simulation like in Modelsim/Questasim/GHDL. Here You will observe an "U" for the sequential logic functions at start-up. I'm open for discussion on how it "should" be.

maehne, 2025-05-27:
I agree with @BFH-ktt1 that the current behaviour is in line with the name of the configuration option “Memory elements start in unknown/random state (rather than 0)” and behaviour, which you would observe in HDL simulation. Therefore, I don't see a need for changing this behaviour. In case, random state is wished as an option, the configuration option should be changed to allow three choices for the memory state to be either zero, unknown, or random. Personally, I would prefer "unknown" to be the default state to match HDL simulation.

prlaba, 2025-05-27:
Adding a third option, "random", when configuring a memory component would certainly meet my requirements.

BFH-ktt1, 2025-05-30:
PR's in this direction are welcomed. Closing this issue as it is not a bug, but desired behavior on the given preset.

closed 2025-05-30 by BFH-ktt1