logisim-evolution/logisim-evolution #2267: POR's output does not return to 0 when used in subcircuit

prlaba, 2025-06-20:
When the simulation is initialized or reset, the POR component's output is 1 (bright green) for the specified duration, then changes to 0 (dark green).

But if you create a subcircuit with the POR component, the subcircuit's output on the main circuit changes to 1 (bright-green) for the specified duration, but never changes back to 0 (dark green). Any inputs tied to the subcircuit's output also remain 1, and any wires tied to the subcircuit's output also remain bright green.

If you then click on *any* trace connected to the subcircut's output, the output immediately changes to 0 (dark green) and any inputs tied to the POR's output also immediately change to 0.

To reproduce:

- Drag the POR component onto the main circuit.  Connect a probe to the POR's output with a short wire.

- Reset the simulation (Ctrl-R).  You should see the probe change to 1 and the wire change to bright green for 2 secs, then change to 0 and dark green.

- Create a subcircuit containing the POR component and a single output (RESET).

- Drag the subcircuit into the main circuit.  Connect a probe to the subcircuit's RESET output with a short wire.

- Reset the simulation (Ctrl-R).  You should see the probe change to 1 and the wire change to bright green for 2 secs, but remain 1 and bright green (it should change to 0 and dark green).

- Click on the wire connecting the probe to the subcircuit's RESET output. The wire immediately changes to dark green and the probe changes to 0.

maehne, 2025-06-20:
Thanks for the report even though it is a duplicate of issue #2111, which is already fixed in our `main` branch. You can use our [nightly builds](https://github.com/logisim-evolution/logisim-evolution?tab=readme-ov-file#nightly-builds) or [build Logisim-evolution from the sources](https://github.com/logisim-evolution/logisim-evolution/blob/main/docs/developers.md#building-from-sources) to profit from the fix and many other improvements.

Feel free to reopen the issue if the nightly builds should not fix your issue. In that case, please provide exact information about your platform and version of Logisim-evolution with which the issue is reproducible.

closed 2025-06-20 by maehne

prlaba, 2025-06-21:
The bug exists in 3.9.0.  Is this bug fixed in a later build?

davidhutchens, 2025-06-21:
@prlaba: Yes, it is fixed in a later build. @maehne gave you a link to the documentation for how to use the nightly build.