logisim-evolution/logisim-evolution #2300: Feature Request: VGA RGB Screen Component with Parallel Pixel Support

Goku144, 2025-08-04:
🚩 Feature Request: VGA RGB Screen Component with Parallel Pixel Support
👋 Hello Logisim Evolution team,

First, thank you for this amazing project! 🙏 I've been using Logisim Evolution to build and simulate digital systems, and I truly appreciate the effort and polish you've put into it.

However, I’ve encountered some limitations with the display components, especially when working on video/gpu-style circuits.

🧱 Current Limitation
The only screen-like component available is the Matrix Panel, which unfortunately:

🟨 Has very low resolution (blocky, oversized pixels)

❌ Lacks proper RGB input (no true color or full-color addressing)

🚫 Is not suitable for realistic VGA or video output simulation

💡 What’s Missing / Feature Request
We would benefit a lot from a more capable VGA screen component with features like:

✅ RGB Inputs:

Separate Red, Green, and Blue pins (1-bit or multi-bit per channel)

Or packed RGB bus

✅ Configurable resolution:

e.g. 320×240, 640×480, or even custom (user-settable)

✅ Control Signals:

Optionally support HSYNC/VSYNC for VGA-like timing

✅ Parallel Pixel Input:

Allow writing multiple pixels per tick (parallel rendering)

Useful for simulating framebuffer writes or GPU-style buses

🎯 Why This Matters
This would make Logisim Evolution far more powerful for educational and experimental GPU/video system design — especially for students and hobbyists working on:

🖼️ Custom video signal generators

🎮 Basic GPU/PPU designs

💾 Memory-mapped video output

🙏 Closing
Would this kind of feature be considered for a future release?
Or is there any workaround to simulate similar behavior with current components?

Thanks again for your awesome work! 💙

Best regards,
[Anonymous]

maehne, 2025-08-25:
I think you overlooked the components "Input/Output > RGB Video" and "System On a Chip > VGA screen", which at least cover some of your requirements mentioned above. There have been some improvements to the "RGB Video" component recently. So make sure to have a look at our [Nightly builds](https://github.com/logisim-evolution/logisim-evolution#nightly-builds). The component is still [limited regarding the selectable screen sizes being powers of 2 and not supporting common aspect ratios](https://github.com/logisim-evolution/logisim-evolution/pull/2288#issuecomment-3108661098). PRs to address this issue in a backwards-compatible way are welcome. @crossplatformdev might be interested to help.

Considering HSYNC and VSYNC as well as multiple pixel per tick assignments would require new models. Personally, I think that a multiple pixel write capability is less essential and within scope of the project.

Enhancements in this area are of low priority for us, as we think such video applications are better dealt with in advanced courses where the students already have switched to using a proper HDL.

maehne, 2025-08-25:
Also issue #2292 is related to this issue.