logisim-evolution/logisim-evolution #2301: 【Problem】Construct NAND gate  using transistors

wellhaomin, 2025-08-05:
This is a simple problem: I want to construct a NAND gate using a P-mos and N-mos , a source and a ground.
When the input is 0 ,the output is 1 ,it is accurate,but the input is 1 the output is U, Why???, Is it because of compatibility issues？

[image]

[image]

davidhutchens, 2025-08-05:
In case someone else reads this, as @wellhaomin figured out, the ^ directional indicator in the transistor gives the direction of information flow, not current flow. So the lower transistor needs to be flipped from south to north.

wellhaomin, 2025-08-08:
> In case someone else reads this, as [@wellhaomin](https://github.com/wellhaomin) figured out, the ^ directional indicator in the transistor gives the direction of information flow, not current flow. So the lower transistor needs to be flipped from south to north.如果其他人读到这篇文章，正如所弄清楚的那样，晶体管中的 ^ 方向指示器给出了信息流的方向，而不是电流流。所以下层晶体管需要从南向北翻转。

Thank you so much！