logisim-evolution/logisim-evolution #2340: New simulation core causes the SOC-simulator to stop working correctly at high tick-frequencies

BFH-ktt1, 2025-09-15:
When setting for example the tick frequency to 2048 kHz, one cannot insert correctly bus-transactions manually into the bus. The exact cause I do not know, I have to look into the details

davidhutchens, 2025-09-20:
I think I have some idea of what is happening. It may take a while for me to verify it, figure out the extent of the issue, and propose a fix. It seems to be related to the new simulation engine reusing InstanceStateImpl objects in the simulation.

maehne, 2025-09-29:
Thanks @davidhutchens for your hard work trying to understand the cause of this issue and finding a fix, which doesn't impact performance! We look forward to your PR to review and test it!

BFH-ktt1, 2025-10-03:
Hello @davidhutchens,
Sorry for the delay. Attached two programs to test, being a "hello world" and a bit more bigger program "bouncing ball".

[programms.zip](https://github.com/user-attachments/files/22674254/programms.zip)