logisim-evolution/logisim-evolution #2351: Menu rendering offset on KDE Wayland

lucas-strummer, 2025-09-25:
### Bug report

Hi,  
I'm experiencing an issue with **Logisim-Evolution** on versions 3.9.0 and 4.0.  

**Environment:** KDE Plasma + Wayland  
**Problem:** dropdown menus are rendered far away from their expected position.  

This happens on the `.deb` releases I tested, both on Debian and Kubuntu, across 3 different PCs.  

Here are two screenshots (but the problem occurs with every menu):  

[image]
[image]

Thanks in advance. Let me know if you need more info for further diagnosis.  

---

### System information

- **OS:** Kubuntu x86_64  
- **Kernel:** Linux 6.14.0-32-generic  
- **DE:** KDE Plasma 6.3.4  
- **WM:** KWin (Wayland)  
- **Theme/Icons:** Breeze Dark  
- **Display setup:**  
  - 1920x1080 @ 75 Hz (24", external)  
  - 1366x768 @ 60 Hz (19", external)  
- **GPU:** AMD Device 7590 (Discrete)  
- **CPU:** AMD Ryzen 7 8700F (16) @ 5.05 GHz  
- **Memory:** 31 GiB

BFH-ktt1, 2025-10-01:
As we use the standard Java functions, I have the feeling this has to do with java interacting with Wayland (I might be wrong), and that it is not a problem of Logisim-Evolution; please also post this issue on the java-community and Kubuntu-community. As I do not have a machine to reproduce I cannot debug :-<

maehne, 2025-12-29:
I am closing this issue due to missing feedback. Providing DEB packages for Debian is tracked in issue #1569. Feel free to reopen if the issue persists with `openjdk-21-jdk` or newer.

closed 2025-12-29 by maehne